valterc/ki2 #249: New system update

CamJones64, 2024-10-19:
After my ride today received a notice there is a new system update to load, 1.513.1975. Has anyone loaded this yet? If so, any issues. Update notes indicates can centre align labels as well as data in the data fields, as well as fixing a few things that cause the unit to crash. 

After our most recent experience not rushing in the load new updates.

CamJones64, 2024-10-20:
Closed as found previous query on same topic in closed issues section.

closed 2024-10-20 by CamJones64

valterc, 2024-10-20:
Thanks, everything is indeed working fine with this new version. Feel free to upgrade 🙂

steve-scott, 2024-10-20:
Thanks for confirming @valterc . I finally updated everything today. Took a few reboots to get the Hammerhead up, but then updating Ki2 was easy. Thanks for all the efforts on this! Enjoy the coffee... :-)

valterc, 2024-10-20:
Thanks Steve! The kind words and the coffee are very much appreciated 🙂