valterc/ki2 #250: OS update 1.516.1983

CamJones64, 2024-11-02:
Has anyone loaded this update as yet? Any issues? 

Gee, I have become a bit gun shy after all the issues with 1.509.1960.

alibikestuff, 2024-11-02:
Cam, I was running 1.513.1975 and was going to report a new issue here:
Although Hammerhead had fixed the issue whereby, a ride that was resumed after a power down (e.g. at a lunch stop) would contiue the ride, but would NOT display the climber during the resumed part of the ride, it seems that they have broken Ki2 again with this change !

I found that after resuming the ride, although my ride continue as expected and Climber did appear, the Ki2 app could no longer connect to my di2, it would simply display NA for all Ki2 fields, such as front and rear gear. I was able to verify this as a resume issue, by stopping and saving the ride and starting a new ride, Ki2 was working perfectly again.

After returning from the ride, I saw and installed the latest HH update (1.516.1983) but I am a little concerned, as they have put a whole load of changes in regarding battery savings options and customisation of those settings, so I am kind of expecting them to have broken something else.

I feel like they should commit to only once monthly updates and perform a decent amount of regression testing in that time !

I wasn't planning on another ride on this bike, in the next day or two, but if I get a chance I will have a look at how this behaves.

CamJones64, 2024-11-02:
@alibikestuff thanks for that. The update notes from HH indicate the battery options relate to the new Karoo unit. There is also a note that they had fixed an issue with climber but was related to rerouting.

I know when I stop during a ride, I pause the unit and let it go into sleep mode rather than power down, so I have not experienced an issue with losing my Di2 connection. Am riding again tomorrow so might try the power down approach and see if it affects Di2 like you are observing.

Will only perform the update after the ride so if there are any issues, hopefully someone might have reported them by then.

proseLA, 2024-12-21:
has anyone updated to this version on a karoo 2?  is ki2 working with it?

thanks in advance.

CamJones64, 2024-12-22:
> has anyone updated to this version on a karoo 2? is ki2 working with it?
> 
> thanks in advance.

Yep, and have upgraded to further versions since. All is working as it should.

CamJones64, 2025-01-13:
> Closing this issue. Recommendation is to eventually update (manually) to
> Ki2 version 10.
>
> *Important note for previous Ki2 users:* This new Ki2 version, using
> Karoo Extensions, is not backwards compatible, old Ki2 versions need to be
> uninstalled and this new version installed from scratch. Existing ride
> elements from previous Ki2 versions will not work with the new version -
> all ride profiles must be updated.
>
> You may choose to keep using the old Ki2 version, but there will be no
> support for that version going forward. Apologies for that, but Hammerhead
> simply deprecated the old Karoo SDK and the new Karoo Extensions are built
> quite differently. This 'reset' is not ideal as it requires some manual
> work from Ki2 users, but it is the best way to move forward. Hammerhead
> will only officially support Karoo Extensions so it does not make sense for
> me to keep using the old and rusty Karoo SDK. Fully switching to Karoo
> Extensions gives us more confidence that things will not break in future
> (hopefully).

Valter,

Can we load the new version of Ki2 onto the K2 head unit, or do we need to
stay with the older version? If we do need to migrate to the new Ki2
version, is there any specific instructions we need to be aware of to
remove the old version from our K2 units?

Thanks

Cameron

valterc, 2025-01-13:
You can use the new Ki2 version in Karoo 2. You will need to manually uninstall the old version and then install the new version. The new versions are **not** backwards compatible, you will also need to **re-configure Karoo ride profiles** and any settings in Ki2.

General guide to install the new version:

1. Find the link to the Ki2 APK file from the latest [release page](https://github.com/valterc/ki2/releases/latest/)
2. [Sideload Ki2 into Karoo](https://support.hammerhead.io/hc/en-us/articles/31576497036827-Companion-App-Sideloading) (For Karoo 2 follow this [video](https://www.youtube.com/watch?v=qp7H_ZPQEJY))
3. [Connect to shifting devices from Ki2](https://user-images.githubusercontent.com/1299179/204136334-c8a5a395-c6b5-4d16-a8d5-ff1fa2dd726d.mp4)
   - Open Ki2
   - Press the `+` button
   - Make sure a wireless shifting device is nearby and that it is active (perform a shift or press a button to wake up the shifting system)
   - Press the `Add` button in Ki2 to add a device once it is found
4. Configure the shifting device from Karoo (for example: rename the device - optional)
5. Add the Ki2 devices to Karoo
   - Open Karoo sensors
   - Press the `+` button to add a new sensor
   - Choose the `Extensions` sensor type (the icon with the puzzle piece)
   - Select all shifting connections that were previously added in Ki2 (no scan for new devices is possible here, you must add the device first in Ki2)
   - Configure gearing information in Karoo sensors
6. Use default Karoo shifting elements or Ki2 elements in your ride profile
7. Customize battery alerts, shifting audio alerts, button controls, overlays and other features from Ki2