valterc/ki2 #251: Ki2 with STEPS / EP8

philvdch, 2024-11-04:
Hello,
I installed the app on my Karoo2 and I can only to read the % of the motor battery. Can I read other value as power ?

Thanks for you work. Phil

valterc, 2024-11-05:
Hi! 

Unfortunately not currently. It should be possible but someone needs to decode and interpret the values that the STEPS broadcasts. I don't have a STEPS bike, so I'm not able to do that myself. 

If you have programming experience feel free to give it a go and submit your findings.

Favez, 2025-01-13:
> Hi! I am a new owner of a Karoo and I have a steps system and I am a developer (dotnet). Please give me a hint how you trace broadcast logs, then I will give it a try.

That would be so awesome!
Maybe you should have a look at the ST Ride app (https://www.emtbforums.com/threads/st-ride-app-tracker-dash-logger-for-steps-motors.9357/)
This app is able to pull loads of information from the EP8 motor, almost everything you can wish for. The developer is active in the thread.

mweihberg, 2025-01-14:
From what i quickly read about this ST Ride App:
- they grap the information via bluetooth not ant+ (ant+ ist possible for external HRM sensors)
- He says: Especially SC-E8000-Displays are not compatible due to a lack of telemetry data via bluetooth

i ve also the owner of a Suunto Race and they offer a Shimano Steps Integration via Suunto Plus Store. It works flawless and i can obtain any data i want. https://forum.suunto.com/topic/10702/shimano-e-bike-systems/50

So the main question is if how for e.g. Suunto can optain data from the SC-E8000 display? How do they obtain the data from bluetooth?

kszaboa, 2026-09-13:
and I have an SC-E7000 display on my Steps E8000 bike and use it with ST Ride and ST Unlocker. Can I help? Like logging the ant+signals or whatsoever?

I am more than disappointed in Shimano now, as I switched from my old Garmin Edge explore to Karoo 3 just to realize how petty-minded these japanese are.