valterc/ki2 #266: K3 adding Ki2

maxdc72, 2025-01-05:
Hi there I tried to install the Ki2 on my k3 but I keep getting the install failed (it does download). Any tips?

valterc, 2025-01-05:
Hi, can you let me know where/when do you see the install failed message? What are the steps that you are doing?

Thanks!

maxdc72, 2025-01-05:
Hi thereI lunch the APK right click and share to HH app (iPhone)Max

ringods, 2025-01-06:
I guess this is the same as #269

maxdc72, 2025-01-06:
I don’t think so as I am using the K3 and I uninstalled the old version before installing the new one

valterc, 2025-01-06:
I don't think there is much we can do on my side, my suggestion is to follow the recommendations from Hammerhead and try again.

If you have an old Ki2 version installed, please uninstall it and restart Karoo before attempting to install a new Ki2 version.

[Recommendations and guide from Hammerhead](https://support.hammerhead.io/hc/en-us/articles/31576497036827-Companion-App-Sideloading)
>Prerequisites:
> -  Karoo should be updated to SW Version 1.527.2014 or higher.  
>   -  Android Companion App Version should be 1.36.0 or higher.
>   - iPhone Companion App Version should be 1.12.0 or higher.
> - The Karoo should be connected to Wi-Fi. 
> - The connected phone should have an active Internet connection.

maxdc72, 2025-01-06:
Hi thereI toggled the dev options on and off.  Turned the k3 on and off and then tried again and it worked.  Amazing work thank you for the helpMax

95wildtt, 2025-01-06:
Overlay stays at 0, 5 after setting is changed and conformed in the settings screen.

valterc, 2025-01-07:
Closing this issue. Thanks @95wildtt, created a new issue #270 to track that problem.

closed 2025-01-07 by valterc