valterc/ki2 #267: Update to 8.2

Gekran, 2025-01-05:
Hi Valter, I successfully installed Ki2 version 10 on my Karoo3. But I got the message that there is an update. But that’s an update to 8.2.

valterc, 2025-01-05:
Whoops! Please ignore that 'update', it showed up because Ki2 version 10 was marked as pre-release, so it tried to get the latest non pre-release version. You can ignore that update notification, it will not show up again as I moved version 10 to be the latest official release.

ringods, 2025-01-06:
But marking it as an official release prevents me of updating to 8.x now. I'm still on 7.1 at the moment and the version it will update to is 10. Updating fails with the error message from #269 (which is offcourse logical)

valterc, 2025-01-06:
> But marking it as an official release prevents me of updating to 8.x now. I'm still on 7.1 at the moment and the version it will update to is 10. Updating fails with the error message from #269 (which is offcourse logical)

Yes, unfortunately that's how the update work, it always tries to update to the latest version. Since the new version 10 is not backwards compatible with older versions, you will not be able to update. Instead you'll need to manually uninstall and install the new version. You can choose to remain on version 7.1 if that works for you, but you won't have further support.

> **Important note for previous Ki2 users:** This new Ki2 version, using Karoo Extensions, is not backwards compatible, old Ki2 versions need to be uninstalled and this new version installed from scratch. Existing ride elements from previous Ki2 versions will not work with the new version - all ride profiles must be updated.

> You may choose to keep using the old Ki2 version, but there will be no support for that version going forward. Apologies for that, but Hammerhead simply deprecated the old Karoo SDK and the new Karoo Extensions are built quite differently. This 'reset' is not ideal as it requires some manual work from Ki2 users, but it is the best way to move forward. Hammerhead will only officially support Karoo Extensions so it does not make sense for me to keep using the old and rusty Karoo SDK. Fully switching to Karoo Extensions gives us more confidence that things will not break in future (hopefully).

closed 2025-01-06 by valterc