valterc/ki2 #269: Unable to update from 8.1 to 10.0

gino3004, 2025-01-06:
Hey Valter tried to install the 10.0 Ki2 version. The Error message I get reds like:
Error:
INSTALL_FAILED_UPDATE-INCOMPATIBLE: Package com.valter.ki2 signatures do not match the previously installed verson;ignoring!

I am using a K2 and have the former versions always installed without any issue, obviously didn't update to 8.2....but want now to skip that one and go t0 10.0
Let me know if I can do anything at my end about it.
Thank you somuch  for your help,
Cheers Gijs Nooteboom
gino300459@gmail.com

valterc, 2025-01-06:
You cannot update from version 8 to the newer version 10. 

That is by design because the newer version is **not** backwards compatible. If you want to update to version 10, you will need to uninstall the old version and then re-install Ki2 version 10 manually.

> **Important note for previous Ki2 users:** This new Ki2 version, using Karoo Extensions, is not backwards compatible, old Ki2 versions need to be uninstalled and this new version installed from scratch. Existing ride elements from previous Ki2 versions will not work with the new version - all ride profiles must be updated.

> You may choose to keep using the old Ki2 version, but there will be no support for that version going forward. Apologies for that, but Hammerhead simply deprecated the old Karoo SDK and the new Karoo Extensions are built quite differently. This 'reset' is not ideal as it requires some manual work from Ki2 users, but it is the best way to move forward. Hammerhead will only officially support Karoo Extensions so it does not make sense for me to keep using the old and rusty Karoo SDK. Fully switching to Karoo Extensions gives us more confidence that things will not break in future (hopefully).

closed 2025-01-06 by valterc

ringods, 2025-01-06:
@valterc on Karoo 2, should insstalling version 10 still be done via side loading?

Asking this because of this:
https://support.hammerhead.io/hc/en-us/community/posts/31988518958619-How-to-load-extensions-on-Karoo-2

valterc, 2025-01-06:
Yes, for Karoo 2 the install procedure is still the same, manually side loading the app. This is not something that we can control from the app side, only Hammerhead can change/make it easier.

gino3004, 2025-01-06:
Thanks Valter,
I hadn’t understood that but it all makes sense now. No worries will keep
the old version going until it breaks! By then uninstall and side load the
new version.
Appreciate your help and time.
Cheers Gijs.

Sent from Gijs’s mobile device