valterc/ki2 #276: Ki2 Update  failed

Matthew-John-McDonald, 2025-01-09:
I have attempted multiple times to install the Ki2 update on my Karoo 2 (from version 8.1 to 10.0) but have been unsuccessful. I have this displayed on my Karoo 2:

Error:
INSTALL_FAILED_UPDATE_INCOMPATIBLE: Package com.valterc.ki2 signatures do not match the previously installed version; ignoring!

Any suggestions?

iggbom, 2025-01-09:
10.0 is not backwards compatible so this is by design. You have to install 10.0 from scratch. See #269

valterc, 2025-01-09:
As @iggbom pointed out you cannot update from a older version to 10.0

That is by design because the newer version is **not** backwards compatible. If you want to use the version 10, you will need to uninstall the old version and then re-install Ki2 version 10 manually.

> **Important note for previous Ki2 users:** This new Ki2 version, using Karoo Extensions, is not backwards compatible, old Ki2 versions need to be uninstalled and this new version installed from scratch. Existing ride elements from previous Ki2 versions will not work with the new version - all ride profiles must be updated.

> You may choose to keep using the old Ki2 version, but there will be no support for that version going forward. Apologies for that, but Hammerhead simply deprecated the old Karoo SDK and the new Karoo Extensions are built quite differently. This 'reset' is not ideal as it requires some manual work from Ki2 users, but it is the best way to move forward. Hammerhead will only officially support Karoo Extensions so it does not make sense for me to keep using the old and rusty Karoo SDK. Fully switching to Karoo Extensions gives us more confidence that things will not break in future (hopefully).

closed 2025-01-09 by valterc