valterc/ki2 #277: Cogs displayed instead pf gears

bassoastra, 2025-01-09:
Thx so much Valterc for pushing your latest Ki2 extension! Everything worked like a breeze, except for the fact that it only shows the cog info instead of the gears. While the correct no pf cogs is displayed, I obviously would prefer to see direct gears. Had no chance to find a way to change. Thank you!

bassoastra, 2025-01-10:
To precise; this issue is with the data field ‚Gears‘, which shows both chainring & casette. When I choose separate data fields such as ‚Front Gear‘ and ‚Rear Gear‘, it acutually shows the corresponding gears, not cogs count.

in the ‚Gears’ data field and the ‚Gears Visual’ graph however, it always displays the cogs.

valterc, 2025-01-10:
Thanks for the feedback 🙂 This is the default behavior from Karoo profile data fields.

I'm planning on adding extra fields to Ki2 itself where you can choose if you want to display the gear index or gear size. Should be available in the next or following Ki2 version.

This work was already described in #272, so I will close this issue.