valterc/ki2 #278: Errors in Adding Ki2 Data fields to Profiles

nickburge93, 2025-01-10:
Hi valterc, 

I have re-sideloaded the latest version of Ki2 (after the Hammerhead SDK update) and immediately noticed that when trying to add Ki2 data fields to my riding profiles that when I toggle the drop down, **_most_** of the data fields are either **not** displaying at all or display **incorrectly** as shown in the image attached. It changes depending on whether you have just selected a drop down from another category eg. 'BODY TEMP', then the names of those data fields end up under the 'Ki2' heading.

[image]

I am curious to hear your thoughts on this and if there are any other user reports of this issue? Thanks.

nickburge93, 2025-01-10:
Upon further inspection, I believe this is an issue with the new SDK and not the Ki2 code (_of course Valter never lets us down_)

See below:

https://support.hammerhead.io/hc/en-us/community/posts/32649390375835-New-Extension-Capabilities-in-Karoo-Release-1-535-2029

_"Please note there is a known issue where the Title does not appear correctly in the data field picker menu for Extension data fields. The issue has been fixed and will be resolved in our next update. "_

So I take it this is a 'wait and see' kind of fix?

valterc, 2025-01-10:
Yes, this is caused by the latest Karoo software update and impacts all extensions, we cannot change/fix anything on Ki2 side. Just need to wait for the next Karoo software release.

The problem is just visual, you can actually add the profile elements by picking the _wrong_ elements below Ki2, the correct element will appear on the top and in the ride screen.

valterc, 2025-01-10:
Only 3 custom data fields are available from Ki2 for now:
- Shifting Battery %
- Shifting Mode
- Gear Ratio

The goal is to **use the native Karoo shifting data fields** (Gears in text and graphically), but since I got several request already, I will add some custom data fields, as described here: #272

nickburge93, 2025-01-10:
Thanks Valter! Clear.

valterc, 2025-01-28:
This is fixed in the latest Karoo software.

closed 2025-01-28 by valterc