valterc/ki2 #288: Small layout change for graphical field "Gears visual view"?

kintrupf, 2025-01-28:
It's pretty obvious that this is a gear display, so the title "Gears" is note really needed 😉
Could you instead display the remaining charge in percent?
The green battery icon to left is much too small to read it accurately. Is this 30%? Or only 15%? Hard to tell...

Maybe it's even possible to combine the data fields "Gears Index (visual)" and "Gears Size (visual)"?
Could you try something like "R02(52) F04(14)" instead of "R02-F04"?

[image]

valterc, 2025-01-28:
That is nice a suggestion.

I will probably leave the existing fields as-is, but I can add a new one based on your suggestions.

The battery is indeed hard to 'read', it changes the value but also changes colors based on the status, for example:

- Green: Battery is above `Low Status`, i.e. nothing to worry about unless you are planning a multi-day away from home ride
- Yellow: Battery is in `Low Status`, i.e. consider charging soon
- Red: Battery is in `Critical Status`, i.e. charge ASAP

(Low and Critical percentage is configurable in Ki2 settings)

valterc, 2025-02-06:
[image]

Something like that?

kintrupf, 2025-02-06:
I really like it 😊 
Thank you!

valterc, 2025-02-06:
Cool! Will be available on the next Ki2 version 🙂

PR #292 (2025-02-06): Add detailed gears datatype
[image]

Fixes #288
closed 2025-02-06 by valterc