valterc/ki2 #289: Feature Request: Clear Overlay with Hood Button

95wildtt, 2025-01-28:
Hi Valter,

Setting up my new Karoo(3), and would like to be able to clear an overlay to see the data hidden behind it with a button press.

I have my overlay set to stay on all of the time, but on occasion I may want to see the data behind it, and touching the screen to clear it is OK, but a long button press would be better. :-)

Any way you can add this?

Thanks.

valterc, 2025-01-28:
Hi! Not currently possible, it is possible to show the overlay from a button press, but not to hide it.

However it makes sense to be able to do that, it is a nice suggestion, I can add this for the next ki2 version. 

I will add two new options:
- Hide overlay
- Toggle overlay

95wildtt, 2025-01-28:
Excellent, thank you.

PR #294 (2025-02-08): Add switch actions to control overlay
Closes #289
closed 2025-02-08 by valterc