valterc/ki2 #291: Future request: button functionality

Marf79, 2025-02-06:
Hi,
would it be possible to switch on/off lights connected via ANT+ (like Garmin UT800 and Varia 515). Just discovered such need while entering unexpectedly long tunnel on a quick descent;)

valterc, 2025-02-06:
Hi, that sure would be possible, but we need Karoo extensions support. 

It's actually something that was already requested in https://github.com/hammerheadnav/karoo-ext/issues/41, so in the future once that feature is available in Karoo Extensions, then we can add support for it in Ki2.

Marf79, 2025-02-10:
Great! Looking forward...

Schoelzchen, 2026-10-30:
I recently bought a new light with an ANT+ connection and searched for a way to control it with Ki2.
There is a repository that uses some undocumented APIs  [https://github.com/JaJaJim/LightONKaroo](url)