valterc/ki2 #295: Shift counts does not exist

hknklbk, 2025-02-09:
Hi,

After installing the latest (version 12.0), I am not able to see Shift Count under ki2 Data Pages tab.

valterc, 2025-02-09:
Hi, you are right. You need to use the one from Karoo, the one in the Shifting menu.

For that one to work you need to have added the Di2 extension device in Karoo devices.

hknklbk, 2025-02-09:
Hi again,
I did it, but it did not work either.
Unfortunately I revert back to version 8.2.

Shifting menu. For that one to work you need to have added the Di2
extension device in Karoo devices.

*M. Hakan KALEBEK*
İplik İşletme Müdürü

*Durkar Halı Tekstil San. ve Tic. A.Ş.*
3. OSB. Mehmet Batallı Blv. 83305. Cd. No: 4 *|* 27600 *|* Baspinar *|*|
Gaziantep |*|*| TURKIYE
*T:* +90 342 337 9390  *|*  *F: *+90 342 337 9395  *| * *M: *+90 532 602
2895

hknklbk, 2025-02-09:
> Hi again,
> I did it, but it did not work either.
> Unfortunately I revert back to version 8.2.
>  Hi, you are right. You need to use the one from Karoo, the one in the
> Shifting menu. For that one to work you need to have added the Di2
> […](#)

Yes, I understand. 
I did it. I selected from Shifting Menu after adding Did, but it showed nothing on driving page.

valterc, 2025-02-09:
> but it showed nothing on driving page

Can you provide more details? What does it show? "No Sensor"?

I would like to avoid replicating the Karoo shift count element, that one should work once the Ki2 device is added to Karoo extension sensors.

Also keep in mind that more features will not work unless you add the Ki2 device in the Karoo **extension** sensors. I will clarify that in the readme.

hknklbk, 2025-02-13:
I have added the Di2 on Ki2.
Then try to activate Shift Count from Karoo2 shift data page, but it does
not show data when riding.

9 Şub 2025 Paz 22:32 tarihinde Valter Costa ***@***.***> şunu
yazdı:

valterc, 2025-02-15:
> I have added the Di2 on Ki2.
>Then try to activate Shift Count from Karoo2 shift data page, but it does not show data when riding.

You also need to add the Ki2 device to Karoo sensors. That is why it doesn't work 🙂 

I will add a shift count element from Ki2 but I won't guarantee it will have the same count as the one from Karoo.

PR #300 (2025-02-15): Add shift count data type
Fixes #295
closed 2025-02-15 by valterc