valterc/ki2 #297: Settings for Overlay information

bigmig-ZA, 2025-02-14:
I cannot select certain features on the overlay of information e.g. POSITION, OPACITY. There is no OK button. 

[image]
[image]

valterc, 2025-02-14:
Hi, can you tell me which hardware (Karoo 2 or Karoo 3) and which Karoo software version do you have?

bigmig-ZA, 2025-02-14:
Hello Valter,

Karoo (3) 1.538.2049.20, version and Ki2 12.0 (30).

valterc, 2025-02-14:
I was able to reproduce this. 

It happens after changing Karoo language, changing it back to English has no difference, all dialog default android strings are messed up. Restarting doesn't fix the problem.

Not sure if we can fix this without some workarounds, seems to be a Karoo bug impacting all android apps.

valterc, 2025-02-14:
Got a fix for this, will be available on the next Ki2 version.

closed 2025-02-14 by valterc

bigmig-ZA, 2025-02-15:
Thank you, Valter. Enjoy your coffee. ;-)

PR #298 (2025-02-14): Fix dialog text after Karoo language change
Fixes #297