valterc/ki2 #307: upgrade from version 8.1 to 13

cervelo45, 2025-02-17:
Hello all
do I have to upgrade to version 13 I am on karoo2 with version 8.1?
at first valterc said that we were not obliged. but today does version 13 bring additional features compared to 8.1

thank you for your feedback

best regards

valterc, 2025-02-17:
Hi, 

It's up to you. If version 8.1 is working fine for you then you can continue to use that version. 

Having said that, my recommendation is to install a newer version because version 8/8.1 will not get any new features or bug fixes. So if anything stops working on version 8.1 you will need to eventually move to a new version.

The newer versions use Karoo extensions, which is the Hammerhead recommended way of building apps for Karoo, making it more future-proof - also compatible with Karoo 3.

The only downside of moving from 8.1 to 13 is that you will need to manually uninstall the old version and then install the new version. You will also need to configure Ki2 and **re-add all Ki2 ride elements to the profiles**.

EddyS33, 2025-02-20:
Is there an 'idiots guide' for how I manually uninstall the old version? Thanks.

valterc, 2025-02-20:
From the Karoo apps screen, press and hold on the Ki2 app, it will open the extensions app with an options to remove/uninstall Ki2.

See more here: https://support.hammerhead.io/hc/en-us/articles/1260804560130-Karoo-2-Deleting-a-3rd-Party-App

EddyS33, 2025-02-20:
> From the Karoo apps screen, press and hold on the Ki2 app, it will open the extensions app with an options to remove/uninstall Ki2.
> 
> See more here: https://support.hammerhead.io/hc/en-us/articles/1260804560130-Karoo-2-Deleting-a-3rd-Party-App

Many thanks for your help/guidance. I had misunderstood and thought I needed to use the SDK tool to do the uninstall.

cervelo45, 2025-02-22:
Hello
did you have a link that explains the installation procedure?

Thanks a lot

EddyS33, 2025-02-22:
> Hello did you have a link that explains the installation procedure?
> 
> Thanks a lot

It's here on the main ki2 page - https://github.com/valterc/ki2?tab=readme-ov-file#how-can-i-install-the-app

valterc, 2025-02-22:
If the instructions on the Readme are not enough you can find other guides online, the one from DC Rainmaker is also really good: https://www.dcrainmaker.com/2021/02/how-to-sideload-android-apps-on-your-hammerhead-karoo-1-karoo-2.html

cervelo45, 2025-02-22:
It's good I succeeded
not easy with IOS but it's done

thank a lot

closed 2025-02-22 by valterc