valterc/ki2 #311: Control Karoo screens from the shifting buttons

Sakarovitch, 2025-03-02:
Hello Valterc!
First of all big thanks for this KI2 app, I found it really great to use with my Karoo 2.
For some reason, I can't find the switch option to control the Karoo screen.
I can control the overlay information or go to the map page but can't switch from one page to the other.
Maybe something for the next upgrade if possible.
All in all, very pleased with the Di2 functionality I have been missing for months, specially the baterry level info with could be pretty anoying at some point.

valterc, 2025-03-02:
Hi, thanks for the kind words and great to hear that Ki2 is useful for you.

Karoo screens are controlled via the top left and top right Karoo buttons, if you want to control the screens via Di2 you need to select the "Top Left Button" and "Top Right Button" as the actions for Di2 switches, in the Ki2 switch options.

Give it a try and let me know if that works.

Sakarovitch, 2025-03-03:
Hello Valtrec.

Thanks a lot for your answer, I grandly appreciate it.

I didn’t realize that the ‘’top left button’’ and ‘’top right button’’ in
the Ki2 switch options where referring to the physical buttons of the Karoo
2.

I thought they were referring to the position of the switches on the
handlebar of the bike,…

I will give it a try as soon as I get back home but I’m 100% pretty sure it
will perfectly work.

Big thumbs up for your support!