valterc/ki2 #313: View shift battery %

smithl5, 2025-03-15:
I have the Karoo2 and have been using Ki2 for almost 1 year. There are so many battery fields. I can't find fields that show the battery % for the right and left shifters.

I appreciate all the work you've done on this and helping us Karoo users have a more robust system.

Thanks,
Lori Smith

proseLA, 2025-03-18:
someone please correct me if i am wrong...  but Ki2 is for an unnamed japanese company that makes something called Di2; and that product only has 1 battery that powers 4 different devices including the 2 mentioned in this thread...

if they have changed that, and i have mis-spoken, i apologize; and please someone feel free to correct me.

95wildtt, 2025-03-18:
The latest Di2 has batteries in the shift levers too, so they are wireless...

Oskarsfar, 2025-06-20:
Yes, shifter battery indicators would be a nice addition.