valterc/ki2 #315: Shift Mode change not working with Akroo3

chriswabisabi, 2025-03-28:
Hello and thanks for the great extension!
I bought a karoo 3 last week  and I am trying to figure out how to change shift mode of my di2 while riding.
Karoo version 1.548.2092.20
ki2: 13.0 (31)

Connecting the extension works fine, I also see correct data like gear, battery, and shift mode.
But how do I change the shift mode. 
I read just tapping on the shift mode data field while riding would allow in a popup to change the mode.
Nothings happening here with my karoo. Only long klick leads to the karoo change data field dialogue.
Am i doing something wrong or is this not possible at all?

Thanks!

valterc, 2025-03-28:
Hi, that worked on older Ki2 versions before the Karoo extensions, but currently changing shift mode is only possible from the Ki2 app, on the Di2 details page. 

The ride elements don't support changing the shift mode, actually it's the first time I see anyone requesting/using that feature, that's why I didn't add it for this new version, but I guess it's something that could be added.

chriswabisabi, 2025-03-28:
Thanks!
Well that's fine too. (just read how it seems to not work anymore and I didnt find that info on the manual page or overread it).
I don't switch shift mode that often.

One more problem.
When I boot up Karoo and start riding there seems to be no autoconnect of di2, which means I have to go to the extension page open ki2 and then sesnor connects and the numbers show up.
Is that an intended behaviour?

valterc, 2025-03-28:
Ki2 connects automatically to Di2 if Di2 is awake on the first minutes of riding. If you start riding without shifting or pressing the Di2 buttons for several minutes, then Ki2 cannot connect as Di2 is not broadcasting any data.

chriswabisabi, 2025-04-01:
I tried that but the autoconnect does not work for me. I started the karoo and a new ride and shifting also and within the first 10 minutes nothing showed up.
I had to go to extensions and press the connect button within ki2, then instantly the data was there.

valterc, 2025-04-01:
That can be caused by many things, could even be caused by issues on the latest Karoo software, as others are mentioned here: https://support.hammerhead.io/hc/en-us/community/posts/35217997842331

I'm sorry but in that case you will need to troubleshoot the issue yourself, as there is nothing I can do without access to your setup.

fkintrup, 2025-05-20:
And I was wondering why changing the shift mode in Ki2 did not work on my rental bike 😁
If it's not to much trouble please add this feature.
Ususally it's not needed, but in the case of a rental bike it would be quite useful.

fkintrup, 2025-05-20:
Changing the shift mode in the Ki2 app did not seem to work with the Karoo 3, at least I did not notice any changes in the shifting behavior of the 105 Di2 of the rented bike.

I think changing the shift mode during a ride is seldom needed. It's more like a one-time setup option.