valterc/ki2 #317: Gear Display on Karoo 3 Profile Page

timsidell, 2025-04-12:
Hi,

I'm new to GitHub, Ki2 App and the Karoo 3 so apologies if I'm breaking protocol!

I've just installed Ki2 to my Karoo 3 to integrate with Di2. I was hoping to have the "current gear" permanently displayed within the "Gears" data field in the profile page, but I can't seem to achieve this. Instead, I only see the current gear via the overlay feature when the system is triggered.

Is it possible to view "current gear" within the data field in the profile page?

Thanks!

Tim

valterc, 2025-04-12:
There are Ki2 ride elements that show the current gear.

> I was hoping to have the "current gear" permanently displayed within the "Gears" data field in the profile page, but I can't seem to achieve this

Can you be a bit more specific on what is not working? What happens when you add the Gears element to the ride profile?

Keep in mind that the Karoo native shifting elements only work when you add the Ki2 extension sensor to Karoo sensors. Otherwise you can select the Gears element from the Ki2 section the profile customization.

See: https://github.com/valterc/ki2?tab=readme-ov-file#how-can-i-install-the-app

timsidell, 2025-04-13:
> valterc

Hi,

Thanks for your prompt reply and sorry if I wasn’t clear. 

I was unable to achieve the current gear displayed within Karoo 3’s native profile data as pictured here:
￼
But I now realise it was because I hadn’t succeeded in adding the Di2 as a sensor. I’ve now done so and everything seems to be working perfectly - thanks!

Best wishes,

Tim

valterc, 2025-04-13:
Great! Glad it's working now.

Adding the sensor to Karoo also gives you:

- Native battery notifications when Di2 battery is low
- Gear shifts recorded in the ride FIT file - which you can view in the Hammerhead dashboard or in other ride analysis apps.

closed 2025-04-13 by valterc