valterc/ki2 #319: "Bike Disconnected" warning when not riding Di2 bike.

SamDCFC, 2025-04-22:
When riding my SRAM AXS bike I use a different profile with no Ki2 data fields but the App still seems to constantly search for the Di2 system on my other bike and sporadically pops up with the "Disconnected" warning. Can this be prevented or can the App be enabled by profile only? Or can the App be stopped manually?

Steve-53, 2025-05-03:
I'm also looking for this functionality - I use my Karoo 2 on two different bikes, only one of them has a Di2 groupset. It would indeed be nice if Ki2 could be linked to specific profiles.

Steve-53, 2025-05-08:
That would certainly work for me - I'd be more than willing to rename my profiles....

SamDCFC, 2025-05-08:
The “Force Stop” function that Hammerhead recently added seems to do most
of what I wanted. It’s the “Bike X Disconnected” message that pops up that
bugs me and makes me wonder if battery is being used up by the device
continually searching for the Di2. Now after using the Force Stop it still
seems to pop that message up once at the start but it doesn’t repeat.

Thanks for your help

closed 2026-09-15 by valterc