valterc/ki2 #320: Karoo intermittently doesn't connect to Di2

formula155, 2025-04-22:
Sometimes when I power up the Karoo and my bike the Ki2 app doesn't connect. I tried powering off the Karoo and powering it back on again when this happens, but it does not resolve the issue. At other times I can power the Karoo on and Ki2 connects fine.

Oskarsfar, 2025-04-25:
I have the same issue. I thought i missed an "autostart Ki2 extension" setting somewhere.

I just realize that my Hood buttons doesnt change the screens, but when i then open the extension manually (a bit problematic while riding) i connects and works fine.

iggbom, 2025-04-25:
Same experience here. Most of the time I have to manually first navigate to extensions and open ki2. When I do it pretty much always connects. But if I immediately start a ride it is rarely connected.

pilotguy74, 2025-04-27:
Same here for me for a while now, maybe since Hammerhead did the extensions revamp. If I open ki2 before a ride it connects no problem, however, in the before times it would just connect when I woke up my Japanese shifting gruppo with a shift or 3…

heckcj, 2025-05-01:
I have been experiencing the same issue, but it’s not intermittent at all. I have to open Ki2 on my Karoo (3) and then the application will connect to my Di2. I believe this started with the March 27 Karoo software update. I see they’ve issued another update today, May 1st, but don’t yet know if that will or won’t eliminate this issue.

alibikestuff, 2025-05-03:
> I have been experiencing the same issue, but it’s not intermittent at all. I have to open Ki2 on my Karoo (3) every single ride and then the application will connect to my Di2. It’s not critical in that the app still works fine after it connects, but it is a frustrating nuisance, IMHO. I believe this started with the March 27 Karoo software update. I see they’ve issued another update today, May 1st, but don’t yet know if that will eliminate this issue.

I have the Karro3 and can cofirm that the latest firmware version (1.554.2125) fixes this issue, no longer a need to open Ki2 in extensions to connect to DI2 to see your ki2 data in your ride at runtime.

I do however have issues with the gearing data fields:
Although the 'Gears Index Visual' gearing display IS displayed correctly, the following fields NEVER display any values (they also don't display example values when you select these data fields in a 'Profile' and more importantly they do not display actual values at runtime 
Front Gear
Rear Gear
Front Gear T
Rear Gear T

yetdog, 2025-05-05:
> I have the Karro3 and can cofirm that the latest firmware version (1.554.2125) fixes this issue, no longer a need to open Ki2 in extensions to connect to DI2 to see your ki2 data in your ride at [...]

Thanks for reporting your experience on the topic at hand, but please take the other issue into another existing issue thread or create a new issue.

Can anyone else confirm that the latest OS update solved this issue? I’m always cautious of jumping on updates due to their propensity to break stuff.

heckcj, 2025-05-08:
Unfortunately, the latest update did not correct the problem for me. This morning I went on my first ride since updating and Ki2 did not connect to my Di2, I had to manually start the app for it to discover and connect. I will try uninstalling and reinstalling Ki2 to see if that changes anything and will report back.

valterc, 2025-05-08:
Hi everyone, 

Trying to find a way to reproduce this issue. I haven't experienced this problem.

Which gear information do you all have on your profiles?

- Do you have Ki2 extension sensor added to Karoo?
- Do you use only native Karoo gear elements?
- Do you use only Ki2 gear elements?
- No fields, just Ki2 overlay?

heckcj, 2025-05-08:
I'm not  even uninstall Ki2 from my Karoo to see if a reinstallation eliminates the problem. Clicking on the Uninstall button, even after forcing the application to quit, has no effect. Just to make sure I wasn't doing anything wrong, I uninstalled another extension (downloaded from the Karoo "store") successfully.

valterc, I will try to answer your questions, above. I have Karoo OS Version 1.554.2125.20 and Ki2 13.0 installed. I don't use native Karoo gear elements, I only use the Ki2 gear elements on one data page; the elements I use in fields are: Shifting Battery %, Gear Ratio, and Gear Detailed (Visual).  I also use one Ki2 overlay with Dark - Default (gear index).

RustedRoot, 2025-05-08:
Valter, in order of your questions, my answers:

-Yes, Ki2 sensor is added
-I've customized my gear elements in Ki2 which I believed carried over to sensor when I activated it. 46-30 front, 11-36 cassette
-Only field is gear elements, e.g., 46-19. No overlay. That's the only field I use in my profile
-Latest K2 version installed

I've previously uninstalled/reinstalled Ki2 to no effect, connection problem remained.

fkintrup, 2025-05-20:
I also have this problem on at least a third of my rides.
Usually I press one of the hood buttons before I power on the Karoo, so the Di2 should be awake. But even if I forget this I also always shift a few times in the first minutes. So Ki2 has a very good chance to find the Di2 wide awake in the first few minutes of each ride.

Most of the time I notice that Ki2 is not connected when the hood buttons do not work as expected. Opening the Ki2 app during the ride always immediatly connects to the Di2. After switching back to the ride profile everything works.

I use the Ki2 overlay and have several data fields with Di2 information (gear display, shift count, battery). However, these data fields are on a secondary page, not on the main map display.
Hoever, I did not add Ki2 as an extension sensor until now. I just didn't know about it.

heckcj, 2025-05-20:
I didn’t pick up on the need to add Ki2 as an extension sensor, either. I just added it and will report back if that solves (fingers crossed) the problem.

punky22, 2025-06-03:
I have been experiencing the same issue during the past 2-3 weeks, although it is not intermittent. If I turn off the Karoo 2 (1.559.2149.12), it never connects to Di2 the next time I start. Before each ride, I have to remember to open the Ki2 extension (v. 13) and manually connect. I push the button on the derailleur to make it discoverable, but I do not know if this is necessary. Then it stays connected throughout the ride. I just use two data fields: Shifting Battery % and Gears. I also depend on the hood buttons to switch screens. A resolution would be greatly appreciated. Otherwise, thank you for the app.

Oskarsfar, 2025-07-03:
@valterc - are you able reproduce the issue, or do you need anything more from this community?

Please note that i am VERY grateful for this great add-on, this is just a little annoying. 

Thank you very much!

RustedRoot, 2025-07-04:
@valterc , I also continue to have the same issue. Would a logcat be useful!

alibikestuff, 2025-07-13:
> I also continue to have this issue. I am running the latest Karoo2 firmware from a few days ago and ki2 13.0. I have to manually open ki2 before (or during a ride) to get it to connect to my Japanese electronic groupset (the one that starts with a “u” 🙂 if that matters). Not sure if the one oh five works better… the ace one should functionally be the same as the ultra one I’d imagine as they share the same guts?

Mine is 105 and still have the same issue.

95wildtt, 2025-07-20:
Also having the issue, as stated have to open the Ki2 extension before it will connect to my Ultegra 8050 GS.

pilotguy74, 2025-07-25:
Cautiously optimistic but two consecutive rides since the latest Karoo2 update yesterday (July 24) and the di2 seems to connect to the head unit without having to manually open the ki2 extension… two more rides planned this weekend. Will continue to monitor. Hopefully this issue is resolved. Would like to hear if it’s working for anyone else in the thread. Fingers crossed. 🤞

95wildtt, 2025-07-26:
I've only done 1 ride since the last HH update, but Ki2 did connected right away... 
The Karoo (3) also seems to be much faster to connect and just generally quicker to perform display updates.

foobarth, 2025-08-06:
>BTW, is there a way to open the Ki2 app to connect without ending the ride?

Sure, no need to quit anything. Just long-press the lower left button to get to the home screen. From there, navigate to extensions and open Ki2 to connect. Then back to the home screen and lower right button to get back into the still running ride.

matejmarti, 2025-09-14:
I noticed that it does not connect automatically with both my Ultegra 8100 Di2 and GRX 825 di2 UNLESS I wake up the di2 before powering on the Karoo (I'm on karoo 3, new karoo, with build 1.581.2228.20 and ki2 version 13)

pilotguy74, 2025-09-14:
> I noticed that it does not connect automatically with both my Ultegra 8100 Di2 and GRX 825 di2 UNLESS I wake up the di2 before powering on the Karoo (I'm on karoo 3, new karoo, with build 1.581.2228.20 and ki2 version 13)

Yes, I have noticed this too. If I wake up my shifters before powering on the head unit, it seems to connect OK 3/4 of the time, but sometimes I still have to open the extension manually. It’s very sporadic.

punky22, 2025-09-14:
> Yes, I have noticed this too. If I wake up my shifters before powering on the head unit, it seems to connect OK 3/4 of the time, but sometimes I still have to open the extension manually. It’s very [...]

My experience as well. Automatic connection is intermittent, so I just try to remember to open app and connect manually before starting my ride. Either way, making a connection requires waking up Di2 by tapping up and down buttons.
.

Kroterfil, 2025-09-25:
If you turn on Karoo and open your profile always work fine. But if you dont open a profile, it wont Connect. This is the case. Hoping a solution.......

95wildtt, 2025-09-30:
Seems to have gotten worse and happens more often after the latest update from HH...

ap857, 2025-12-10:
I am having this issue as well (more often than not). In my case Ki2 does not appear to be making connection with the bike if I perform a typical routine of powering on the computer, waiting for it to boot, and then starting the ride. I see similar behavior whether I have ensured that the drivetrain is awake before booting the computer or not. 

I'm running Karoo build 1.598.2293.20 and Ki2 13.0

I have 4 bikes currently paired. All are 12speed generation components (2x 9250, 1x 8150, and 1x825). All exhibit the same behavior.

- Each of the bikes has a corresponding Ki2 extension sensor provisioned
- I do not use Karoo native gear elements. I only have the extension sensors setup to get shift data into the .fit file
- I only use the Ki2 overlay, none of the Ki2 data fields

ap857, 2025-12-11:
Well, I've been trying to force it to fail for the past day and it has worked perfectly every time. I'll keep trying and will hopefully be able to get something to you soon.

valterc, 2025-12-11:
Thank you. 

If you notice a pattern, like it only fails when you wait a long time or when you open some other extension before or when you choose a particular profile, etc. That could be useful to know as well.

ap857, 2025-12-15:
Had a completely different failure mode yesterday. 

- Went through my normal routine to start the computer, pair the power meter, and do the zero compensation.
- Switched to the ride profile and the "Connecting..." overlay popped up, but ki2 never connected to the bike even after repeatedly operating the shifter buttons. 
- Gave up on waiting for it to work and opened the ki2 extension
- It still would not connect
- Then ki2 froze and I was eventually prompted to kill the process, which I did.
- Then I re-started ki2 and it had lost its configuration (no bikes listed)
- I finally powered down the K3 and restarted it. 
- This time ki2 found the bike immediately and there were no issues (all of the lost configuration items came back after the reboot). 

I was away from home so I wasn't able to pull logs.

Kroterfil, 2026-01-07:
It works for me. I switch off developers options and use it several times and connect inmediatly. Looks it was the problem.....

ap857, 2026-01-08:
In my case the behavior is independent of the developer option setting as well. 

I am finding that I get more predictable behavior if I launch Ki2 after the Karoo powers on and then operate the shifters to get it to connect. This usually results in a working connection in a short period. On occasion, though, I've had to power-cycle the Karoo to get Ki2 to make a connection.

My-Lambo, 2026-08-15:
FYI.  I recently asked Hammerhead if they would supply an earlier software update so I could bring my K3 back to when Ki2 would work.  Their answer was no.  Then they mentioned Ki2 is not theirs so fix in sight.  Hopefully, with mirml”s code suggestions Valter can get Ki2 running again on my K3.

Munka9, 2026-09-04:
I still  think adding a 5 second sleep commend to the start of the processConnections() would ensure the Ant service is ready before the rest of that function is run and solve the issue for 99% of users. Any chance you could try that? Its not the most elegant fix, but it is the simplest to implement! And if it does work would give more weight to Mirml's issue being a correct diagnosis.

mirml, 2026-09-12:
A lot of actions, thanks @valterc 👍

valterc, 2026-09-17:
Give it a try with latest Ki2 version (16.0)

mirml, 2026-09-19:
​I think this issue is fixed!
​I tested it a few times indoors: I turned on the Karoo first and activated the Di2 afterward. Every single time, the connection was established automatically without needing to open the Ki2 extension menu like before.

​It would be great to hear findings from others as well, but on my end, it seems to be working ok. Thanks! 🙂👍

ToranBrades, 2026-09-19:
Just from my side: Everything appears to be working now. I turned the Karoo on and off at least ten times since yesterday, restarting routes or starting fresh - it worked.

Thank you very much!

pilotguy74, 2026-09-19:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Can confirm this is connecting properly again. Tested yesterday by waking
up the di2 before powering on the head unit, and today just by starting a
ride and waking up the shifters with the first shift. Connected both times
correctly.

Karoo3 latest firmware,
Ki2 16.0

Thank you Valter

gm

My-Lambo, 2026-09-19:
I updated the firmware yesterday on my K3 and updated the Ki2 16 software today.  It WORKS!!!  I’m so happy.  You guys rock!!!

Thank you Valter…

Munka9, 2026-10-05:
Thank you, I can also confirm this latest update is a full fix for me too!

valterc, 2026-10-12:
Thanks for updating with the results, closing this issue now 🙂

closed 2026-10-12 by valterc

pilotguy74, 2026-10-12:
🫡 the saga has ended. Nicely done!

gm

alibikestuff, 2026-10-13:
I am also using v16.0 and I agree, ki2 is working fine for the initial connection:
1) power up Karro, start a ride with ki2 data fields (in my case Battery level, shift mode and gears)
2) activate di2 (shift up or down)
- within a couple of seconds, the ki2 connection is made and my ki2 data fields are successfully populate
Note: I also have a Garmin Forerunner 955 watch connected for Heart Rate and a Garmin Varia Radar (watch is still on and monitoring ride and HR)

3) I get to a coffee stop with friends and turn the Garmin Radar off and I also turn the Karro off (with a ride still active)
4) After coffee stop, I turn the Garmin Varia Radar on and power up the Karro
5) I resume the ride on the Karro
- the Garmin Varia radar is connected again and I can see this on the Karro and my HR is also displayed
BUT, the ki2 data fields are never populated/ki2 does not connect to the bike/di2, despite my changing gear soon after starting the Karro again - this is happening for me every time in this scenario.
The only way I can get the ki2 data connection activated again is to End the Ride on the Karro and Start a new Ride, in this case the ki2 data fields are immediately populated.

alibikestuff, 2026-10-13:
Valter, 

I added my findings.

Perhaps my issue is more complex with additional sensors connected, the other users don't mention any additional sensors in their tests.

Thanks
Allison