valterc/ki2 #326: With two Di2 circuits, only the first listed one is displayed in the data field

Attu-pbr, 2025-06-05:
When using 2x Di2 circuits, the different circuits cannot be assigned to the corresponding profiles. 
Current state: 
I have two different bikes. 
Each has a Di2 circuit. 
Unfortunately, the gears can NOT be assigned to the corresponding bikes/profiles. 
Only the circuit that is in the top position in the Ki2 APP is displayed in the data field. 
It doesn't matter if the right bike profile has been chosen. 

The circuit in the Ki2 APP in the 2nd position is never displayed in the data field.

valterc, 2025-06-05:
This is by design, the first Di2 that is connected is the one that will be displayed. 

You can still add multiple Di2 devices, then when you grab a bike, even if that Di2 is the second on the list, the app will connect to that Di2 and will display the correct values - unless you take the two bikes together on the same ride, in that case you will need to enable/disable the Di2 manually to choose which one to view data from.

Attu-pbr, 2025-06-07:
Unfortunately, this does not work as described, because **(in my opinion use K2)** the assignment between Ki2 APP and bike profile is missing/or incorrect. 
Condition: 
I have Di2 circuit #11 and #22. 
In Ki2-APP, they are also listed #11 in the first place and #22 in the second place. 

Bicycle profile "A" = Di2 # 11 Bicycle profile "B" = Di2 # 22. 

If I now ride the bike "B", the connection to # 22 is signaled as connected in the Ki2 app, as is the case with the sensors. 
Only the data fields in profile "B" remain "empty". 

Only when I move the # 22 gearshift to position one in the Ki2 APP, 
the values appear in the bike profile. 

So I am forced to change the order of the circuits in the Ki2 APP when changing profiles.

Gruß Peter

closed 2025-10-17 by valterc